sipeed/MaixPy #9: MaixPy 4.2.0: Wifi Setup without runtime not possible ..

bkircher67, 2024-05-31:
… but runtime installation requires Wifi. I‘m confused.

[image]

Neutree, 2024-06-01:
maybe its a bug, just use the old version, we will fix it netx monday....

if you want to use this image, just connect it by maixvision and edit the `/boot/wifi.ssid` and `wifi.pass` file, and reboot ( it's better to soft reboot by execute `os.system("reboot")`, when bootup, the wifi will automatically connect to your AP

bkircher67, 2024-06-01:
Thanks! I already fixed it with an uart-ssh session and using the wpa_supplicant app to connect via wifi. Then I could install the runtime. But I assume your solution is faster and more elegant …

Neutree, 2024-06-01:
I fiexed it, just try: https://github.com/sipeed/MaixPy/releases

closed 2024-06-04 by Neutree