sipeed/MaixPy #13: program exit failed. exit code: 1. while running custom model yolov5s after run about a minute

JimlyAdam, 2024-06-07:
Module: [MIPI_RX], Build Time[#2 PREEMPT Mon Jun 3 14:54:37 HKT 2024]

------------Combo DEV ATTR--------------
   Devno  WorkMode  DataType   WDRMode         LinkId        PN Swap  SyncMode  DataEndian  SyncCodeEndian
       0      MIPI     RAW10      NONE  4, 3, 2,-1,-1  0, 0, 0, 0, 0       N/A         N/A             N/A

------------MIPI info-------------------
 Devno EccErr CrcErr HdrErr WcErr fifofull   decode
     0      1      1      0     0        0    raw10
  Physical:       D0       D1       D2       D3       D4       D5
                   0        0        0        0        0        0
   Digital:       D0       D1       D2       D3    CK_HS  CK_ULPS  CK_STOP   CK_ERR   Deskew
             hs_idle  hs_idle  hs_idle  hs_idle        1        0        0        0     idle
Traceback (most recent call last):
  File "/tmp/maixpy_run/main.py", line 102, in <module>
    objs = detector.detect(img, conf_th = 0.5, iou_th = 0.45)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: detect(): incompatible function arguments. The following argument types are supported:
    1. (self: maix._maix.nn.YOLOv5, img: maix._maix.image.Image, conf_th: float = 0.5, iou_th: float = 0.45, fit: maix._maix.image.Fit = <Fit.FIT_CONTAIN: 1>) -> List[maix::nn::Object]

Invoked with: <maix._maix.nn.YOLOv5 object at 0x3fd8280070>, None; kwargs: conf_th=0.5, iou_th=0.45
maix multi-media driver destroyed.
ISP Vipipe(0) Free pa(0x8dff7000) va(0x0x3fd503d000)
[VI BE_Dbg_Info]
VIPreBEDoneSts		:0x0		VIPreBEDmaIdleStatus	:0x7
[VI Post_Dbg_Info]
VIIspTopStatus		:0x3ff
[VI DMA_Dbg_Info]
VIWdma0ErrStatus	:0x3000000	VIWdma0IdleStatus	:0xffffffff
VIWdma1ErrStatus	:0x3000000	VIWdma1IdleStatus	:0xffffffff
VIRdmaErrStatus		:0x3000000	VIRdmaIdleStatus	:0xffffffff
[VI ISP_PIPE_A FE_Dbg_Info]
VIPreFERawDbgSts	:0x3ff		VIPreFEDbgInfo		:0x3f
[VI ISP_PIPE_A]
VIOutImgWidth		:2560
VIOutImgHeight		:1440
VIInImgWidth		:2560
VIInImgHeight		:1440
VIDevFPS		:  13
VIFPS			:   0
VISofCh0Cnt		:1364
VIPreFECh0Cnt		:1335
VIPreBECh0Cnt		:1335
VIPostCnt		:1335
VIDropCnt		:   0
VIDumpCnt		:   0
[VI ISP_PIPE_A Csi_Dbg_Info]
VICsiIntStatus0		:0x3
VICsiIntStatus1		:0x0
VICsiCh0Dbg		:0x0
VICsiCh1Dbg		:0x0
VICsiOverFlowCnt	:   0
VICsiCh0WidthGTCnt	:   0
VICsiCh0WidthLSCnt	:   0
VICsiCh0HeightGTCnt	:   0
VICsiCh0HeightLSCnt	:  22

program exit failed. exit code: 1.

lxowalle, 2024-06-11:
When reading the camera image determine that cam.read() is not None before processing. By the way can you post your test code, we didn't notice that cam.read() returns None after running for a while, I'm going to try to reproduce it to see

Neutree, 2024-06-19:
v4.3.0 resolved this issue

closed 2024-06-19 by Neutree