sipeed/MaixPy #20: 关于rtmp分辨率问题

junchao98, 2024-06-20:
在文档中的rtmp测试代码分辨率为640x480 ，尝试修改为更大的分辨率例如1920x1080，出现错误

参考的测试代码
```
from maix import camera, time, rtmp, image

cam = camera.Camera(640, 480, image.Format.FMT_YVU420SP)

# rtmp://192.168.0.30:1935/live/stream
host = '192.168.0.30'
port = 1935
app = 'live'
stream = 'stream'
bitrate = 1000_000
r = rtmp.Rtmp(host, port, app, stream, bitrate)
r.bind_camera(cam)
r.start()

while True:
    time.sleep(1)
```

报错信息如下，似乎是buffer空间不够。
```
ViPipe:0,===GC4653 1440P 30fps 10bit LINEAR Init OK!===
********************************************************************************
cvi_bin_isp message
gerritId:      97347          commitId:      fc6ce647
md5:           d6db2297ddfd44e8252c1f3f888f47b2
sensorNum      1
sensorName0    4653

PQBIN message
gerritId:
              commitId:      fc6ce647
md5:           d6db2297ddfd44e8252c1f3f888f47b2
sensorNum      1
sensorName0    4653

author:        lxowalle       desc:          gc4653 30fps
createTime:    2024-03-26 13:41:34version:
tool Version:       v3.0.5.48           mode:
********************************************************************************
[SAMPLE_COMM_ISP_Thread]-95: ISP Dev 0 running!
VI_SDK_IOC_S_CTRL - vi_sdk_enable_chn NG, No buffer space available
ISP Vipipe(0) Free pa(0x8f50a000) va(0x0x3ff1a25000)
-- [E] Trigger signal, code:SIGSEGV(11)!
```

在rtsp 测试用例中分辨率可以为2560x1440。请问rtmp分辨率的问题是软件bug还是受限于硬件资源

lxowalle, 2024-06-20:
这个问题像是同时运行了两个APP,maixcam不允许多个app同时跑. rtmp是支持1920x1080的, 如果还是不行可以升级到最新的系统再尝试

closed 2024-06-20 by junchao98

junchao98, 2024-06-20:
收到，谢谢解答