sipeed/MaixPy #22: RunTimeError : Value is null and camera suddenly shutdown while using yolov8n

Diqtra, 2024-06-25:
[image]
My Maixcam has an error when using yolov8n, and the message says, "RuntimeError: Value is Null: Camera read failed." I'm using the yolov8n that comes with MaixPy. The camera works but only for a little while before stopping.

lxowalle, 2024-06-26:
Hi, from the logs it looks like you're launching multiple apps, please make sure that the MaixVision app and the MaixCAM app aren't launching at the same time (MaixCAM black screen means no apps are running) . If that doesn't solve the problem, you can try to get the latest image from release, update it and try again!

JimlyAdam, 2024-06-26:
It also happened to me, Maixvision app was just for knowing the log file, the fact is the program  was installed and then exit with code 1... This happened even with the latest MaixPy...

[image]

lxowalle, 2024-06-26:
This may not be a software problem, but the camera is not connected properly, you can open the back cover to check if the camera cable is loose.

JimlyAdam, 2024-06-26:
Unfortunately, I have two MaixCAM, but it happened at both, I've tried to replace the MicroSD and check all the flexible but those was safe

closed 2024-09-03 by Diqtra