sipeed/MaixPy #27: 不存在image.send_to_maixvision()

Jerry1962325, 2024-07-02:
我遵循教程，但是发现 send_to_maixvision() 方法不存在，如下两图所示：（正确的图片效果来自我的修复）
[image]
[image]
这或许有用，但是和教程的精神不符：
```Python
from maix import image, display

img = image.Image(320, 240)
disp = display.Display()

img.draw_rect(0, 0, img.width(), img.height(), color=image.Color.from_rgb(255, 0, 0), thickness=-1)
img.draw_rect(10, 10, 100, 100, color=image.Color.from_rgb(255, 0, 0))
img.draw_string(10, 10, "Hello MaixPy!", color=image.Color.from_rgb(255, 255, 255))
while 1:
    disp.show(img)
```

Neutree, 2024-07-02:
噢 文档写错了， 是 `maix.display.send_to_maixvision(img)` 欢迎 PR 修改文档

Jerry1962325, 2024-07-02:
那4.3.2版本的“Fix Bugs”日志也错了2333

closed 2024-07-04 by Neutree