sipeed/MaixPy #29: 可以播放录音后的pcm文件，但无法播放wav音频

mcpanl, 2024-07-04:
可以播放录音后的pcm文件，但无法播放wav音频，代码没报错，就是没有声音。请问需要wav设置特定的比特率才可以正常播放吗？如果是的话，比特率最大为多少？
[image]

lxowalle, 2024-07-05:
1. 请确保更新到了最新的运行库，方法Settings->Install Runtime->Upgrade
2. 请看下面代码的注释说明有PCM参数 “采样率48000，FMT_S16_LE格式，和采样通道数为1”

mcpanl, 2024-07-05:
好的，晚上回家我试一下用4.8k采样率的音频试一下

nis243, 2024-09-19:
Hello.
I want to know how to easily play multiple wav files.
environment: maixcam-2024-09-05-maixpy-v4.5.1.img & Install Runtime 1.19.0

When using multiple audio.Players, the error "Unknown error: capture init failed" occurs.
So I converted the wav to PCM.
   sox 0.wav  -c 1 -r 48000 -t s16 0.pcm
   sox 1.wav  -c 1 -r 48000 -t s16 1.pcm
   sox 2.wav  -c 1 -r 48000 -t s16 2.pcm

`from maix import audio, time, app

p               =   audio.Player()
with open('/root/lets/0.pcm', 'rb') as f:
    ctx_0 = f.read()
with open('/root/lets/1.pcm', 'rb') as f:
    ctx_1 = f.read()
with open('/root/lets/2.pcm', 'rb') as f:
    ctx_2 = f.read()
sounds  =   [ctx_0,ctx_1,ctx_2]
print("sample_rate:{} format:{} channel:{}".format(p.sample_rate(), p.format(), p.channel()))
p.play(bytes(sounds[0]))
c   =   0
while not app.need_exit():
    c   +=  1
    time.sleep_ms(10)
    if  c > 250:
        break
p.play(bytes(sounds[1]))
c   =   0
while not app.need_exit():
    c   +=  1
    time.sleep_ms(10)
    if  c > 250:
        break
p.play(bytes(sounds[2]))
c   =   0
while not app.need_exit():
    c   +=  1
    time.sleep_ms(10)
    if  c > 250:
        break
`

Even when playback is complete, the termination condition is not met, so we are forced to add a timeout process.

If the wait time is too long, the following error will occur:
>alsa pcm wait timeout!
>-- [E] play data length is incorrect, write 163876 bytes, returns 0 bytes

Currently, even after playback is complete, I can't get out of the while loop, which is very frustrating.

lxowalle, 2024-09-19:
> Hello. I want to know how to easily play multiple wav files. environment: maixcam-2024-09-05-maixpy-v4.5.1.img & Install Runtime 1.19.0
> 
> When using multiple audio.Players, the error "Unknown error: capture init failed" occurs. So I converted the wav to PCM. sox 0.wav -c 1 -r 48000 -t s16 0.pcm sox 1.wav -c 1 -r 48000 -t s16 1.pcm sox 2.wav -c 1 -r 48000 -t s16 2.pcm
> 
> `from maix import audio, time, app
> 
> p = audio.Player() with open('/root/lets/0.pcm', 'rb') as f: ctx_0 = f.read() with open('/root/lets/1.pcm', 'rb') as f: ctx_1 = f.read() with open('/root/lets/2.pcm', 'rb') as f: ctx_2 = f.read() sounds = [ctx_0,ctx_1,ctx_2] print("sample_rate:{} format:{} channel:{}".format(p.sample_rate(), p.format(), p.channel())) p.play(bytes(sounds[0])) c = 0 while not app.need_exit(): c += 1 time.sleep_ms(10) if c > 250: break p.play(bytes(sounds[1])) c = 0 while not app.need_exit(): c += 1 time.sleep_ms(10) if c > 250: break p.play(bytes(sounds[2])) c = 0 while not app.need_exit(): c += 1 time.sleep_ms(10) if c > 250: break `
> 
> Even when playback is complete, the termination condition is not met, so we are forced to add a timeout process.
> 
> If the wait time is too long, the following error will occur:
> 
> > alsa pcm wait timeout!
> > -- [E] play data length is incorrect, write 163876 bytes, returns 0 bytes
> 
> Currently, even after playback is complete, I can't get out of the while loop, which is very frustrating.
> 
> The same goes for wav files; even after playback is complete, the while loop does not exit, which is very frustrating.
> 
> What I want to do is play the inference results as audio.

Hello, currently, audio playback is set to non-blocking by default to prevent it from blocking the execution of other code. However, it seems that this might make the code more complex. We will optimize this issue in the next update.

mcpanl, 2024-10-05:
问题已解决，目前可以正常播放wav音频了！

closed 2024-10-05 by mcpanl