sipeed/MaixPy #44: ModuleNotFoundError: No module named 'Maix'

nguyenvulong, 2024-09-05:
I found this documentation:[ https://wiki.sipeed.com/soft/maixpy/en/develop_kit_board/module_microphone.html]( https://wiki.sipeed.com/soft/maixpy/en/develop_kit_board/module_microphone.html)
but it seems like there's no implementation of this module

is it only available in the older version of `MaixPy`?

```python
>>> from Maix import MIC_ARRAY as mic
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'Maix'
>>> from maix import MIC_ARRAY as mic
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'MIC_ARRAY' from 'maix' (/home/lvnguyen/.pyenv/versions/sound-augmentation/lib/python3.9/site-packages/maix/__init__.py)
```

Neutree, 2024-09-05:
this repo is MaixPy v4, not fully compatible with the maixpy v1, the compatible API is here: https://github.com/sipeed/MaixPy/tree/main/maix/v1

Neutree, 2024-09-05:
and all maixpy v4's doc under  https://wiki.sipeed.com/maixpy/   not  https://wiki.sipeed.com/soft/

nguyenvulong, 2024-09-05:
> this repo is MaixPy v4, not fully compatible with the maixpy v1, the compatible API is here: https://github.com/sipeed/MaixPy/tree/main/maix/v1

I believe that it's not implemented there but a placeholder only

It should be here [https://github.com/sipeed/MaixPy-v1](https://github.com/sipeed/MaixPy-v1)

[image]

Neutree, 2024-09-05:
what's your board? MaixCAM or k210, if K210 just use MaixPy v1, if MaixCAM please refer to https://wiki.sipeed.com/maixpy/doc/en/audio/play.html, and MaixCAM not support the  MIC array cause MaixCAM's I2S not suuport yet

closed 2024-09-05 by Neutree

nguyenvulong, 2024-09-05:
I am considering both K210 and MaixCAM
Does `MaixCAM` support [MaixPy-v1](https://github.com/sipeed/MaixPy-v1)?

Thanks.

Neutree, 2024-09-05:
MaixCAM not support MaixPy-v1，but MaixPy v4 contain maybe 95% + MaixPy v1's function and have more ability, if you use vision functions please use MaixCAM.

nguyenvulong, 2024-09-05:
I want to use Audio (for example, sound localization).
I guess I will choose K210 for now.

Good day Neutree!