sipeed/MaixPy #45: MaixCAM如何连接开放式WiFi？

China-Mr-Hou, 2024-09-06:
如题，我想让MaixCAM连接一个没有密码的开放式WiFi，在设置App当中扫描并选择了相应SSID，密码留空，点击连接之后提示保存成功，但IP始终没有显示。

是不是MaixCAM通过WPA的密码认证方式连接开放WiFi导致的？有没有什么好的解决方案（指成功连接上开放式WiFi）？

本人尝试过wpa_cli，肯定不成功（因为wpa_cli好像就是WPA认证专用的），固件本身似乎不自带nmcli

求指点，通过命令行配制的方法也可以接受

Neutree, 2024-09-09:
`/boot/wpa_supplicant.conf` 下面
```
network={
    ssid="你的开放网络SSID"
    key_mgmt=NONE
}
```

应该就行了，开机 /etc/init.d 下面的wifi脚本会读取/boot 下的配置这个文件拷贝到 /etc/wpa_supplicant.conf 然后启动的

China-Mr-Hou, 2024-09-13:
您好，感谢您的回复。
我尝试了您说的方法，但似乎连接不上
除此之外我还尝试了直接将conf文件覆盖/etc目录下的conf，似乎也不起作用
我查看了init.d内部的服务文件，但我感觉Linux启动过程中并没有复制conf文件，即使conf文件存在于boot
我在配置过程中删除了boot中的wifi.ssid wifi.pass和wifi.sta

More help is needed, please :(

Neutree, 2024-09-13:
wifi 的脚本在  `/etc/init.d/S30wifi`，你可以仔细阅读此脚本看到wifi连接的所有行为，开机确实会自动拷贝`/boot/wpa_supplicant.conf`到 `/etc/`目录并且会忽略`/boot/wifi.ssid`和`/boot/wifi.pass`.

China-Mr-Hou, 2024-09-13:
我有查看过S30wifi脚本，里面确实有您描述的代码功能，但我在/boot新建并放好了wpa_supplicant.conf后，/etc目录下的conf文件内容没有变化，如下
`ctrl_interface=/var/run/wpa_supplicant.conf`
然后我就手动拷贝覆盖/etc/wpa_supplicant.conf文件，重启之后还是没有IP

Neutree, 2024-09-13:
> `/boot/wpa_supplicant.conf` 下面
> 
> ```
> network={
>     ssid="你的开放网络SSID"
>     key_mgmt=NONE
> }
> ```
> 
> 应该就行了，开机 /etc/init.d 下面的wifi脚本会读取/boot 下的配置这个文件拷贝到 /etc/wpa_supplicant.conf 然后启动的

你有按照我说的写文件内容吗？贴出你的配置文件内容

China-Mr-Hou, 2024-09-13:
[image]

China-Mr-Hou, 2024-09-13:
已经解决了，我又重新看了S30wifi服务文件脚本，我不应该删除'/boot/wifi.sta'，我又重新创建了这个文件，已经成功连接

closed 2024-09-13 by China-Mr-Hou