sipeed/MaixPy #50: MaixVision pp_ocr.mud   not found

winitboonkliang, 2024-09-30:
Test OCR function issue
maixvision pp_ocr.mud   not found
how to download this file? url

start running...

-- [I] find board: maixcam
[SAMPLE_COMM_SNS_ParseIni]-2185: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1778: devNum =  1
[parse_sensor_name]-1859: sensor =  GCORE_GC4653_MIPI_4M_30FPS_10BIT
[parse_sensor_busid]-1888: bus_id =  4
[parse_sensor_i2caddr]-1899: sns_i2c_addr =  29
[parse_sensor_mipidev]-1910: mipi_dev =  0
[parse_sensor_laneid]-1921: Lane_id =  4, 3, 2, -1, -1
[parse_sensor_pnswap]-1932: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1954: mclk_en =  1
[parse_sensor_mclk]-1965: mclk =  1
try release sys ok
[_mmf_init]-1144: maix multi-media version:67a90e2-musl_riscv64
maix multi-media init ok
try release vio ok
try release venc ok
[SAMPLE_COMM_VO_StartChn]-351: u32Width:368, u32Height:552, u32Square:1
-- [E] model path /root/models/pp_ocr.mud not exists

Traceback (most recent call last):
  File "/tmp/maixpy_run/main.py", line 39, in <module>
    main(screen)
  File "/tmp/maixpy_run/main.py", line 9, in main
    ocr = nn.PP_OCR(model)
          ^^^^^^^^^^^^^^^^
RuntimeError: Invalid arguments: load model failed

winitboonkliang, 2024-09-30:
fix issue with download new firmware last version 4.6
os and file install factory old version

Neutree, 2024-10-08:
burn 4.6 system, the files integrated in latest system, or manually download models from [maixhub.com](https://maixhub.com/model/zoo/449)

closed 2024-10-08 by Neutree