sipeed/MaixPy #58: 自己造好的whl轮子为什么说该平台不支持？

lfpu, 2024-10-24:
import os

os.system('pip install /root/Mywhl/Myexample-1.0-cp311-cp311-win_amd64.whl')

print("done")

下面是报错：

ERROR: Myexample-1.0-cp311-cp311-win_amd64.whl is not a supported wheel on this platform.
WARNING: There was an error checking the latest version of pip.
done

program exited

lfpu, 2024-10-24:
按照[编译 MaixPy 源码](https://wiki.sipeed.com/maixpy/doc/zh/source_code/build.html) 步骤编译轮子报错，详情看下图

[image]

Neutree, 2024-10-24:
Myexample-1.0-cp311-cp311-win_amd64.whl  你读一读文件名， win_amd64 即给 windows x86 编译的，你拿到maixcam 的 risc-v 能运行就怪了。

用 `python setup.py bdist_wheel maixcam` 命令

lfpu, 2024-10-25:
> Myexample-1.0-cp311-cp311-win_amd64.whl 你读一读文件名， win_amd64 即给 windows x86 编译的，你拿到maixcam 的 risc-v 能运行就怪了。
> 
> 用 `python setup.py bdist_wheel maixcam` 命令

噢噢噢噢，多谢

closed 2024-10-25 by lfpu