sipeed/MaixPy #59: 为什么编译MaixPy报错 找不到 'maixtool.app_deploy'

lfpu, 2024-10-25:
[lfpu@localhost MaixPy]$ python setup.py bdist_wheel maixcam
-- SDK_PATH:/home/lfpu/Maix/MaixCDK
Traceback (most recent call last):
  File "/home/lfpu/Maix/MaixPy/project.py", line 67, in <module>
    main()
  File "/home/lfpu/Maix/MaixPy/project.py", line 64, in main
    exec_project_py()
  File "/home/lfpu/Maix/MaixPy/project.py", line 61, in exec_project_py
    main(sdk_path, project_path)
  File "/home/lfpu/Maix/MaixCDK/tools/cmake/project.py", line 283, in main
    extra_tools = get_extra_cmds(sdk_path)
                  ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lfpu/Maix/MaixCDK/tools/cmake/project.py", line 178, in get_extra_cmds
    tool = __import__(name)
           ^^^^^^^^^^^^^^^^
  File "/home/lfpu/Maix/MaixCDK/tools/cmds/deploy.py", line 12, in <module>
    from maixtool.app_deploy import serve
ModuleNotFoundError: No module named 'maixtool.app_deploy'
-- Build cpp modules failed!

Neutree, 2024-10-25:
pip install -U maixtool

lfpu, 2024-10-25:
> pip install -U maixtool

谢谢，是这个问题。 但是，我自己造的whl子不依赖maipy，如何加入到设备呢？官方文档没有答案。

Neutree, 2024-10-25:
纯python编写的可以直接用，用c编写的用指定的工具链编译就好了如 https://wiki.sipeed.com/maixpy/doc/zh/source_code/add_c_module.html#%E9%80%9A%E7%94%A8%E5%87%BD%E6%95%B0%E5%B0%81%E8%A3%85 （写得比较简陋，欢迎PR） 所说， 工具链和编译参数参考 https://github.com/sipeed/MaixCDK/blob/main/platforms/maixcam.yaml

closed 2024-10-26 by Neutree