sipeed/MaixPy #62: 编译报错自定义类的成员函数 识别成普通函数

lfpu, 2024-10-28:
**### 头文件**
#pragma once

/**
* @brief maix.myclass module, test my class
* @maixpy maix.myclass

*/

namespace maix::myclass
{

  class MyClass{

public:
/**
*@brief initialize
*@param a int a
*@param b int b
*/
  MyClass(int a,int b);
  ~MyClass();
    /**
     * My function, add two integer.
     * @return int type, will a + b
     * @maixpy maix.myclass.MyClass.Summrize
     */
    int Summrize();
  private:
  int A; int B;
  };

} // namespace maix::myclass

**### CPP 文件**

/**
 * @file maix_myclass.hpp
 * @brief Maix myclass module test
 * @maixpy maix.myclass
 * @author lfpu

*/

#include "maix_myclass.hpp"

namespace maix::myclass
{
   MyClass::MyClass(int a,int b){
     A=a;
     B=b;
   }   
 int MyClass::Summrize(){
   return A+B;
  }

} // namespace maix::myclass 

**### 报错：**
[image]

Neutree, 2024-10-28:
class MyClass{ 没加注释

lfpu, 2024-10-29:
> class MyClass{ 没加注释

大佬，加了注释，然后编译过了，但是调用不了哦。

### **调用**
from maix import display, myclass

MY=myclass.MyClass(1,2)

print(MY.Summrize())

disp=display.Display()

disp.GetInfo()

### **IDE调试报错：**

start running...

Traceback (most recent call last):
  File "/tmp/maixpy_run/main.py", line 1, in <module>
    from maix import display, myclass
ImportError: cannot import name 'myclass' from 'maix' (/usr/lib/python3.11/site-packages/maix/__init__.py)

program exit failed. exit code: 1. 

### **头文件添加的注释如下**
[image]