sipeed/MaixPy #69: 关于GC4653传感器光学中心问题，以及camera是否支持画面偏移功能呢，目前需要先read为image再进行偏移

mcpanl, 2024-11-08:
Camera是否支持画面偏移功能呢，目前需要先read为image再进行裁剪偏移。

[image]

如上图文档所示，由于GC4653光学中心和芯片中心不在同一个位置，导致安装鱼眼镜头时能看出明显的偏移（X方向偏移量较大，Y方向偏移量较少）。（如下图实拍所示，蓝色箭头才是M12镜头座的中心位置，但它并没有位于传感器的中心）

[image]

目前解决方案是先将cam的画面读取出来，用image的crop()方法进行裁剪，以实现各个方向看起来比较均衡，不过如果能给camera加上一个原生的画面偏移就更好了，这样录制视频之类的就不用先转为image对象了。

另外建议修改一下GC4653芯片的位置，将光学中心和M12镜头座中心对齐，以便优化下一批GC4653镜头模组体验，感谢！另外建议预留一小部分不封胶的模组，便于更换不同高度的镜头座或者带有IR-CUT功能的镜头座，再次感谢！

Neutree, 2024-11-08:
不封胶的模组商业应该可以定制，我问了硬件设计是以光学中心对齐的，有可能是封胶误差？

Neutree, 2024-11-08:
硬件上估计就是比较难做到完全精确的，软件上理论上在摄像头出图流程中加一个可选的crop环节也可以实现

Neutree, 2024-11-08:
camera 的 set_windowing 函数可以实现指定crop，你试试

closed 2024-11-13 by mcpanl