sipeed/MaixPy #73: MaixCAM-Pro中使用audio.Player()播放完wav音频后会有200ms左右的电流声，如何优化或者改善这种现象呢

mcpanl, 2024-11-13:
MaixCAM-Pro中使用audio.Player()播放完wav音频后会有200ms左右的电流声，如何优化或者改善这种现象呢

lxowalle, 2024-11-14:
试试一下检查一下音频数据是不是本身有噪音，或者主动把音频末尾200ms左右的数据置0看看能不能改善。
ps:现在maxicam pro上功放的en脚是直连的3.3v，有动手能力可以把en脚飞线到gpio，播放完成后拉低en脚可以保证静音。

mcpanl, 2024-11-14:
确定不是音频的问题，其实没200ms那么夸张，就是音频播放结束后会有滋滋滋的短暂声响，有时候长按电源键关机也会听到类似的声音。如果研发下一个版本的PCB时候，建议预留EN脚

closed 2025-03-09 by mcpanl

China-Mr-Hou, 2025-03-19:
> MaixCAM-Pro中使用audio.Player()播放完wav音频后会有200ms左右的电流声，如何优化或者改善这种现象呢
> 
> ```
> from maix import audio, time, app
> 
> p = audio.Player("/root/output.wav")
> 
> p.play()
> 
> while not app.need_exit():
>     time.sleep_ms(10)
> print("play finish!")
> ```
> 
> 采样率是48k，采样格式为小端格式-有符号16位，采样通道为1

我在使用MaixCAM Pro调用MaixPy V4文档中关于播放wav的代码，运行到p=audio.Player('/root/filename.wav')时卡住，我后续通过SSH CLI手动运行每一行，是一样的结果，等待了至少5分钟无果，文件很小，时长只有7秒，格式与上述一致。

我在IDE结束该程序（点击红色正方形停止按钮）后，再次运行到定义Player时直接报错cannot open PCM，届时设置内调节音量滑块无效，拖动滑块后顶部显示的百分比不变，随后设置APP闪退并返回退出代码1

请问这是什么问题，如何正确播放音频？

感谢SiPEED团队！

Owen Hou

lxowalle, 2025-03-19:
> > MaixCAM-Pro中使用audio.Player()播放完wav音频后会有200ms左右的当前声，如何优化或者改善这种现象呢
> > ```
> > from maix import audio, time, app
> > 
> > p = audio.Player("/root/output.wav")
> > 
> > p.play()
> > 
> > while not app.need_exit():
> >     time.sleep_ms(10)
> > print("play finish!")
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 采样率为48k，采样格式为小端格式-有符号16位，采样通道为1
> 
> 我在使用MaixCAM Pro调用MaixPy V4文档中关于播放wav的代码，运行到p=audio.Player('/root/filename.wav')时卡住，我后续通过SSH CLI手动运行每行，是一样的结果，等待了至少5分钟无果，文件数量，时长只有7，格式与上述一致。
> 
> 我在IDE结束该程序（点击红色指示灯停止按钮）后，再次运行到定义播放器时直接报错无法打开PCM，届时将设置内部调节音量闹钟无效，拖动后顶部显示的百分点停止按钮，并设置APP闪退并返回退出代码1
> 
> 请问这是什么问题，如何正确播放音频？
> 
> 感谢SiPEED团队！
> 
> 侯先生

这个问题的根源是内核dma内存池上的bug，已经在v4.10.3版本中修复，重新烧录镜像后再试试。

China-Mr-Hou, 2025-03-19:
> 这个问题的根源是内核dma内存池上的bug，已经在v4.10.3版本中修复，重新烧录镜像后再试试。

@lxowalle 看到Release刚刚发布的更新了，我今天尝试以下，感谢您和SiPEED团队！

Owen Hou.