sipeed/MaixPy #88: ImportError: Error loading shared library libpostproc.so.55: No such file or directory

pauceano, 2024-12-06:
I am sorry if this is not the way to do it, but last issue was closed by mistake and I cannot afford to loose this problem, I am in a hurry with this.
Using Maixvision with a MaixCAM:
Trying to run the examples, I get this error: ImportError: Error loading shared library libpostproc.so.55: No such file or directory
I have installed ffmpeg for python, but I do not know how to install libraries and this library is not there.

Thank you for your help

Neutree, 2024-12-09:
I closed it, not mistake, you can install by `sudo apt install libpostproc55`.
**And we not ensure running on PC, we only ensure everything will be okeay on MaixCAM**

closed 2024-12-09 by Neutree

pauceano, 2024-12-09:
Thanks for that. Please, clarify this point. I am running Maixvision with a MaixCAM, following what I think are official instructions. Does it mean Maixvision is not supported? 
I do not want to run things to experimental, if Maixvision is not supported, what is supported environment for testing?
Thanks again.

Neutree, 2024-12-09:
dumplicated with #87