sipeed/MaixPy #94: Can't use video frames with MaixPy due to unsupported format

hellvesper, 2024-12-17:
The problem I thinks with unsupported YUV420 frames format/

I'm trying to use short video fragment with algorithms from MaixPy, I'm getting images from video according to `video.Decoder` example:
```python
d = video.Decoder('/root/output.mp4')
d.seek(0)
ctx = d.decode_video()
img = ctx.image()
```
But images encoded in `<Format.FMT_YVU420SP: 8>` format, and looks like it unsupported by `Image.to_format()`. When I call `img.find_edges(edge_type, threshold=[50, 100])` I'm getting error:
```
Traceback (most recent call last):
  File "/tmp/maixpy_run/main.py", line 44, in <module>
    img.find_edges(edge_type, threshold=[50, 100])
RuntimeError: Not implemented: not support format
```

I've implemented workaround through OpenCV api like this:
```python
img_cv = image.image2cv(img, ensure_bgr=False, copy=False)
# YUV420 to Grayscale
img_gray = cv2.cvtColor(img_cv, cv2.COLOR_YUV2GRAY_I420)
img2 = image.cv2image(img_gray)
disp.show(img2)
```
But performance isn't great. Is there any direct method or solution without opencv?

lxowalle, 2024-12-18:
Right now we don't support NV21 to GRAY, we'll try to support this feature in the new version released this week.

hellvesper, 2024-12-18:
Thanks! Will wait for update

closed 2024-12-18 by hellvesper