sipeed/MaixPy #98: 系统编译错误，有遇到同样问题的吗？

CoboLinux, 2024-12-27:
PART OFFSET: 16777728
+ /home/njzy/sipeed/maix/MaixPy/tools/os/fuse2fs -o fakeroot -o offset=16777728 /home/njzy/sipeed/maix/MaixPy/tools/os/tmp/maixcam-pro-2024-12-27-maixpy-v4.8.2.img img_root
/home/njzy/sipeed/maix/MaixPy/tools/os/tmp/maixcam-pro-2024-12-27-maixpy-v4.8.2.img：不支持写入日志。
fuse: mountpoint is not empty
fuse: if you are sure this is safe, use the 'nonempty' mount option
+ echo 'copy root files now'
copy root files now
+ find tmp/sys_builtin_files -mindepth 1 -maxdepth 1 -type d '!' -name boot
+ read -r dir
+ cp -r tmp/sys_builtin_files/root img_root
+ read -r dir
+ cp -r tmp/sys_builtin_files/usr img_root
+ read -r dir
+ cp -r tmp/sys_builtin_files/maixapp img_root
+ read -r dir
+ cp -r tmp/sys_builtin_files/etc img_root
+ read -r dir
+ sync
+ echo 'copy root files done'
copy root files done
+ umount img_root
umount: img_root: not mounted.
[image]

Neutree, 2024-12-27:
umount失败了，文件夹手动点击弹出虚拟磁盘，再重试

CoboLinux, 2024-12-27:
没有挂载虚拟磁盘啊，求指教，这个问题困扰好几天了

CoboLinux, 2024-12-27:
没有挂载的磁盘啊，困扰好几天了，请指教。
[image]

Neutree, 2024-12-27:
`df -h` 看，也可以问问GPT。
还有就是你发的截图都是没有发成功的

CoboLinux, 2024-12-27:
[image]
请问这条编译系统的命令有问题吗？
./gen_os.sh '/home/njzy/sipeed/LicheeRV-Nano-Build/install/soc_sg2002_licheervnano_sd/images/2024-12-20-17-31-1fcc2f.img' /home/njzy/sipeed/maix/MaixCDK/dist/MaixPy-4.8.2-py3-none-any.whl '/home/njzy/sipeed/maix/MaixPy/sys_builtin_files' 1 maixcam-pro

Neutree, 2024-12-30:
都是在github上的，国内需要终端开代理才能下得动

CoboLinux, 2024-12-30:
> 都是在github上的，国内需要终端开代理才能下得动
用了科学上网也下载不了，不信你试试

Neutree, 2024-12-31:
可以的，你的网络问题，在线action也有测试过的，https://github.com/sipeed/MaixPy/actions

closed 2024-12-31 by Neutree