sipeed/MaixPy #113: 请问再做rtsp 推流的时候如何同时做识别

wwmoo7, 2025-02-24:
我需要在做rtsp 推流的时候同时做识别， 但是RTSP 只支持NV21， 模型的输入又都是RGB的， 如果我自己用CPU 直接转，应该很慢， 有什么推荐的方法吗

lxowalle, 2025-02-25:
camera有个add_channel方法，可以新增一个camera通道，一个通道出rgb，一个通道出nv21。需要注意他们最好在一个线程内操作，不然可能影响取图效率

wwmoo7, 2025-02-26:
感谢回复， 我这样写 cam1 = camera.add_channel(320, 240, image.Format.FMT_RGB888) 提示没有add_channel这个方法 AttributeError: module 'maix._maix.camera' has no attribute 'add_channel'

wwmoo7, 2025-02-26:
cam = camera.Camera(1280, 720, image.Format.FMT_YVU420SP)
server = rtsp.Rtsp()
server.bind_camera(cam)
server.start()
disp = display.Display()

detector = nn.YOLO11(model="/root/models/yolo11n.mud", dual_buff=True)
cam2 = cam.add_channel(detector.input_width(), detector.input_height(), detector.input_format())
这样可以了，目前满足需求了

closed 2025-02-26 by wwmoo7